nkolban/duktape-esp32 #1: simplify curl-esp32 component.mk

morganrallen, 2016-12-01:
According to `esp-idf/docs/build_system.rst`

- ``COMPONENT_PATH``: The component directory. Evaluates to the absolute path of the directory containing ``component.mk``. The component path cannot contain spaces.

`CURL_PATH=$(COMPONENT_PATH)` will work instead

cmason1978, 2016-12-10:
If you don't mind, how are you configuring curl-esp32?

After making the above change I still get this error:

```
/usr/local/src/esp-idf/examples/duktape-esp32/components/curl-esp32/lib/curl_setup.h:132:28: fatal error: curl/curlbuild.h: No such file or directory
 #include <curl/curlbuild.h>
> find . -name curlbuild\*
./components/curl-esp32/include/curl/curlbuild.h.cmake
./components/curl-esp32/include/curl/curlbuild.h.dist
./components/curl-esp32/include/curl/curlbuild.h.in
```

Are you running configure in curl-esp32 to produce the config file? 

Thanks @morganrallen, @nkolban, this looks really interesting/promising!

nkolban, 2016-12-10:
Howdy guys ... sorry for the delay in responding here ... I hadn't realized anyone was looking at this repository yet as the project is still in VERY early days.  The problem has been resolved ... ESP32-Duktape no longer depends on CURL.  In fact even Mongoose will be disappearing soon.

closed 2016-12-10 by nkolban

Phando, 2017-02-21:
@morganrallen 
How did the culr-esp work out for you? Where did you put your curl-esp directory? Where did you put CURL_PATH=$(COMPONENT_PATH)?