nkolban/duktape-esp32 #2: Request: commit messages & feature branches

negativekelvin, 2016-12-21:
Would be great if you could consider descriptive commit messages so it is easier to follow along and if you are adding or reworking a significant feature consider using branch and merge. Thanks.

nkolban, 2016-12-21:
Howdy sir ... I hadn't realized that ANYONE other than me was working on the source at the moment.  Can you ping me on `IRC #ESP32` and we can chat.  Right now this is basically a "sandbox" as so much is changing all over the place on a daily basis.  If there is anyone other than me working on the source then I'll immediately change to issue and tracking basis .... but as of right now, I believe I am the only one "hacking away" and hence am touching all the files all the time.

negativekelvin, 2016-12-22:
Well I am not currently working on it per se but doing some testing and watching.  If it is too disruptive to your development process now then no problem.  But perhaps more organization and annotation will help you too in the long run. Thanks