nkolban/duktape-esp32 #4: alloc failed after http.js

edge0701, 2017-01-18:
Builds fine, but get alloc error on run. New to this so unsure if my device (ESP32 Thing) has less memory than other boards? Here is the relevant section of the boot log:

D (4402) log: Module: require("bootwifi.js") loading "bootwifi.js"
D (4406) dukf_utils: >> dukf_loadFile: (ESPFS) bootwifi.js, heapSize=30652
D (4412) espfs: >> espFsOpen: bootwifi.js
D (4415) dukf_utils: << duk_loadFile: Read file bootwifi.js for size 5556
D (4645) log: Module: require("nvs.js") loading "nvs.js"
D (4648) dukf_utils: >> dukf_loadFile: (ESPFS) nvs.js, heapSize=21260
D (4648) espfs: >> espFsOpen: nvs.js
D (4650) dukf_utils: << duk_loadFile: Read file nvs.js for size 1298
D (4741) modules: >> js_esp32_getNativeFunction
D (4742) modules: - nativeFunctionId that we are looking for is "ModuleNVS"
D (4742) modules: << js_esp32_getNativeFunction
D (4752) log: Module: require("http.js") loading "http.js"
D (4754) dukf_utils: >> dukf_loadFile: (ESPFS) http.js, heapSize=17800
D (4757) espfs: >> espFsOpen: http.js
D (4761) dukf_utils: << duk_loadFile: Read file http.js for size 8449
D (4799) duktape_utils: Error: Error: alloc failed
line: 29
Stack: Error: alloc failed
    at [anon] (/home/edge/src/esp32/duktape-esp32/components/duktape/src/duktape.c:20666) internal
    at require () native strict preventsyield
    at bootwifi.js (bootwifi.js:29) preventsyield
    at require () native strict preventsyield
    at [anon] (start.js:57)
    at global (start.js:64) preventsyield
D (4824) dukf_utils: << dukf_runFile: start.js
D (4828) duktape_task: Free heap at start of JavaScript main loop: 18620
D (4834) log: Console:
 ______  _____ _____ ____ ___
|  ____|/ ____|  __ \___ \__ \
| |__  | (___ | |__) |__) | ) |
|  __|  \___ \|  ___/|__ < / /
| |____ ____) | |    ___) / /_
|______|_____/|_|  _|____/____|
|  __ \      | |  | |
| |  | |_   _| | _| |_ __ _ _ __   ___
| |  | | | | | |/ / __/ _` | '_ \ / _ \
| |__| | |_| |   <| || (_| | |_) |  __/
|_____/ \__,_|_|\_\\__\__,_| .__/ \___|
                           | |
                           |_|
 http://duktape.org
 ESP32 port/framework: Neil Kolban
 Build: Jan 18 2017 12:53:56

D (4883) duktape_task: Starting main loop!

nkolban, 2017-01-18:
Howdy Sir,
Excellent ... please allow me to work with you to get you up and running.  In its default mode, ESP32-Duktape connects to an access point and then becomes a web server serving up editor web pages.  While nice and slick, it is eating RAM.  In the current state of the ESP32, we only have about 200K of RAM to play with.  Subtract WiFi, sockets, HTTP server and a bunch of other bits ... and that RAM starts to get low.

There are TWO immediate solutions:

1) Switch off being a Web Server.  In this mode, we can use a command line too to send JS scripts to ESP32-Duktape over serial for execution.  This means that we don't need to immediately connect to WiFi nor to we need to run a Web Server.

2) Switch off debugging data collection.  If you are building from source (excellent) then the default is to have Duktape collect diagnostic information ... this means that for every JavaScript object and variable created, a ton of stuff is kept so we can see line numbers, human readable variable names and more in the event of an exception in JS.   We can switch that off.

Within the source is a file called ./data/duktape/ESP32-Duktape.yaml

Comment out (by placing a hash at the start) the following:
```
DUK_USE_VERBOSE_ERRORS: true
DUK_USE_TRACEBACKS: true
DUK_USE_AUGMENT_ERROR_CREATE: true
DUK_USE_FUNC_FILENAME_PROPERTY: true
DUK_USE_PC2LINE: true
```

and then run:

```
$ make duktape_configure
$ make clean
$ make flashall
```

Please get back to me with how this works out for you ... and you can find me on `#ESP32` on IRC if I can be of live assistance.

edge0701, 2017-01-18:
And howdy to you, good Sir. Excellent work (from what I can see - I just want to run it now!).
I commented out the aforementioned lines, cleaned the project and reconfigured duktape. I receive a different error message now: 

```
D (4589) log: Module: require("http.js") loading "http.js"
D (4591) dukf_utils: >> dukf_loadFile: (ESPFS) http.js, heapSize=20952
D (4595) espfs: >> espFsOpen: http.js
D (4599) dukf_utils: << duk_loadFile: Read file http.js for size 8449
D (4660) duktape_utils: Error: Error: NULL
line: 0
Stack: Error: 1
D (4660) dukf_utils: << dukf_runFile: start.js
D (4660) duktape_task: Free heap at start of JavaScript main loop: 15724
D (4666) log: Console:
```

nkolban, 2017-01-18:
Unfortunately, things have gone south after the latest ESP-IDF releases.  The amount of free RAM has decreased so much that I can no longer host the internal Web Server and the WiFi stack.  I have a solution which involves sending JavaScript programs to run over serial... but right now it requires a second serial port.  Do you happen to have a second USB->UART adapter that could be plugged into UART1 of the ESP32?

nkolban, 2017-01-18:
Aha!!  I found some extra RAM by disabling WebSocket support.  I've just pushed a new series of commits into GitHub ... can we try again?  If we are still having problems, feel free to email me at kolban1@kolban.com and, if you are willing, we can try some direct support through Skype or IRC

nkolban, 2017-01-20:
Let me know if this specific issue is good ... and if so, go ahead and close it out ... we can always open up new issues as needed.

edge0701, 2017-01-20:
Thank you for working with me to resolve the issues I was having. All is good. Closed.

closed 2017-01-20 by edge0701