nkolban/duktape-esp32 #5: Installation

davidmoshal, 2017-01-18:
just got the sparkfun esp32 board, tried:

```bash
esptool.py \
--chip esp32 \
--port "tty.usbserial-DN02MZEH" \
--baud 115200  \
write_flash \
-z  \
--flash_mode "dio" \
--flash_freq "40m"  \
--flash_size "2MB"  \
0x1000 bootloader.bin 0x10000 \
app-template.bin \
0x8000 partitions_singleapp.bin
```
and got:
```bash
usage: esptool [-h] [--port PORT] [--baud BAUD]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash,verify_flash,erase_flash}
               ...
esptool: error: argument operation: invalid choice: 'esp32' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash', 'verify_flash', 'erase_flash')
```

note: the board and port work fine with the arduino-esp32 libraries.

nkolban, 2017-01-18:
Assuming you are using the binary download of ESP32-Duktape, there should be a shell script called "install_binaries.sh" which contains the command necessary for flashing the ESP32 with duktape.  The copy I see shows:

```
#!/bin/bash
#
#  Script to install the ESP32-Duktape images into an ESP32.
#
#  For faster transfers that *may* be less reliable, change the baud
# rate to 921600.
#

PORT="/dev/ttyUSB0"
BAUD="115200"

set -e

if [ ! -z "$IDF_PATH" ]
then
	ESPTOOL=$IDF_PATH/components/esptool_py/esptool/esptool.py
fi

if [ -z "$ESPTOOL" ]
then
	echo "Could not find esptool.  Make sure that IDF_PATH is set"
	echo "to the root of your ESP-IDF install"
	exit 1
fi

python $ESPTOOL  \
 --chip esp32 --port "$PORT" --baud $BAUD  write_flash --flash_mode "dio" \
 --flash_freq "40m" --flash_size "4MB" --compress \
 0x1000   bootloader.bin \
 0x8000   partitions_singleapp.bin \
 0x10000  esp32-duktape.bin \
 0x180000 spiffs.img \
 0x360000 espfs.img

```

When I run `esptool -h` I see the following:

```
usage: esptool [-h] [--chip {auto,esp8266,esp31,esp32}] [--port PORT]
               [--baud BAUD] [--before {default_reset,no_reset,esp32r0}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...

esptool.py v2.0-dev - ESP8266 ROM Bootloader Utility

positional arguments:
  {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
                        Run esptool {command} -h for additional help
    load_ram            Download an image to RAM and execute
    dump_mem            Dump arbitrary memory to disk
    read_mem            Read arbitrary memory location
    write_mem           Read-modify-write to arbitrary memory location
    write_flash         Write a binary blob to flash
    run                 Run application code in flash
    image_info          Dump headers from an application image
    make_image          Create an application image from binary files
    elf2image           Create an application image from ELF file
    read_mac            Read MAC address from OTP ROM
    chip_id             Read Chip ID from OTP ROM
    flash_id            Read SPI flash manufacturer and device ID
    read_flash_status   Read SPI flash status register
    write_flash_status  Write SPI flash status register
    read_flash          Read SPI flash content
    verify_flash        Verify a binary blob against flash
    erase_flash         Perform Chip Erase on SPI flash
    erase_region        Erase a region of the flash
    version             Print esptool version

optional arguments:
  -h, --help            show this help message and exit
  --chip {auto,esp8266,esp31,esp32}, -c {auto,esp8266,esp31,esp32}
                        Target chip type
  --port PORT, -p PORT  Serial port device
  --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading
  --before {default_reset,no_reset,esp32r0}
                        What to do before connecting to the chip
  --after {hard_reset,soft_reset,no_reset}, -a {hard_reset,soft_reset,no_reset}
                        What to do after esptool.py is finished
  --no-stub             Disable launching the flasher stub, only talk to ROM
                        bootloader. Some features will not be available.

```

Note the version number of "v2.0-dev".   Can you check that you are using the latest version of `esptool` as supplied with the ESP-IDF SDK?

davidmoshal, 2017-01-18:
Thanks Neil, I probably should update esptool too, it's probably out of date.
Dave

closed 2017-01-18 by davidmoshal

nkolban, 2017-01-18:
David ... feel free to drop me an email at kolban1@kolban.com ... I'm really keen on ESP32-Duktape and want to chat with users to make sure all is working for you and to hear about areas that can be improved.

davidmoshal, 2017-01-20:
@nkolban SUCCESS!

I spent a little time this morning on the project and managed to flash the ESP32 successfully from OSX!

Steps:
1. installed esp-idf using this page:
http://esp-idf.readthedocs.io/en/latest/macos-setup.html

specifically:
```bash
dowloaded:
https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz

mkdir -p ~/esp
cd ~/esp
tar -xzf ~/Downloads/xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz
export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin

git clone --recursive https://github.com/espressif/esp-idf.git
export IDF_PATH=~/esp/esp-idf
```

2. downloaded binaries from this page:
https://github.com/nkolban/duktape-esp32/blob/master/docs/README.md

and executed this command with the correct port for my machine:
```bash
 $IDF_PATH/components/esptool_py/esptool/esptool.py --chip esp32 \
--port "/dev/tty.usbserial-DN02MZEH" \
--baud 115200  \
write_flash -z \
--flash_mode "dio" \
--flash_freq "40m"  \
--flash_size "2MB" \
  0x1000 bootloader.bin 0x10000 app-template.bin 0x8000 partitions_singleapp.bin
```

### Notes:

1. As you suggested, originally I was using the wrong esptool.py
2. Your page: https://github.com/nkolban/duktape-esp32/blob/master/docs/installation.md
   suggests that these files are needed, yet I was able to flash without them:
   - espfs.img
   - spiffs.img
   so, these instructions may be in conflict:
   - https://github.com/nkolban/duktape-esp32/blob/master/docs/README.md , and
   - https://github.com/nkolban/duktape-esp32/blob/master/docs/installation.md

davidmoshal, 2017-02-13:
Note: the above information is out of date, there is a newer build, 2017-01-20, which has an `./installbinaries.sh` file whith the correct flashing instructions, together with the correct binaries.

the command is now:
```
# change the port to the port identified by: ls /dev | grep tty

$IDF_PATH/components/esptool_py/esptool/esptool.py \
--chip esp32 \
--port "/dev/tty.SLAB_USBtoUART" \
--baud 115200 write_flash \
--flash_mode "dio" \
--flash_freq "40m" \
--flash_size "4MB" \
--compress \
 0x1000    bootloader.bin \
 0x8000    partitions_singleapp.bin \
 0x10000   esp32-duktape.bin \
 0x180000 spiffs.img \
 0x300000 espfs.img
```