nkolban/duktape-esp32 #11: SSID containing space character is not URL-decoded

pepe79, 2017-07-31:
My SSID has a space character, which is not correctly decoded from the http-request after submit.
Looking at the log output, it seems, that the characters are not decoded after retrieving them from the http-request. My SSID is "HAL9000 2.4"
```
D (4133) log: Going to try and connect to AP named "HAL9000+2.4"
D (6141) module_wifi:  - Connecting to access point: "HAL9000+2.4" with "<Password hidden>"
```

I had the same thing in an arduino sketch some day, and fixed it with this code excerpt:
```
void decodeUrlComponent(String* value) {
  value->replace("+", " ");
  value->replace("%21", "!");
  value->replace("%22", "\"");
  value->replace("%23", "#");
  value->replace("%24", "$");
  value->replace("%25", "%");
  value->replace("%26", "&");
  value->replace("%27", "'");
  value->replace("%28", "(");
  value->replace("%29", ")");
  value->replace("%2A", "*");
  value->replace("%2B", "+");
  value->replace("%2C", ",");
  value->replace("%2D", "-");
  value->replace("%2E", ".");
  value->replace("%2F", "/");
  value->replace("%3A", ":");
  value->replace("%3B", ";");
  value->replace("%3C", "<");
  value->replace("%3D", "=");
  value->replace("%3E", ">");
  value->replace("%3F", "?");
  value->replace("%40", "@");
}

```

nkolban, 2017-07-31:
Excellent ... thanks for catching this.  I'll look into it ASAP.  The good news is that large chunks of ESP32-Duktape are themselves written in JavaScript ... so we can leverage the underlying JavaScript APIs to achieve resolution ... for example ... 

https://www.w3schools.com/jsref/jsref_decodeuricomponent.asp

I'll find where that is needed in the code base and affect a fix.

Usage of ESP32-Duktape has been low ... ive found that the amount of RAM on an ESP32 basic isn't enough for the majority of work items ... however with the new WROVER, we now have over 4MBytes.  Do you have a WROVER?

pepe79, 2017-07-31:
No, I have no WROVER, but it looks promising. 

decodeURIcomponent leads to an instruction error in my installation...., but i did not looked into it further.

boneskull, 2018-02-06:
#19 is a dupe of this.  I sent a PR for it, which is #20.

boneskull, 2018-02-06:
the code changes in 9fb31f6 did not work