nkolban/duktape-esp32 #23: how to run tests?

boneskull, 2018-02-06:
There are two separate dirs full of tests; one in `filesystem/tests/` and `tests/`.  I'm not sure how to run them.

dashxdr, 2018-02-11:
They're associated with the ide you just got working... You can click the load button and you'll see them listed.

boneskull, 2018-02-11:
this is what happens when I try to flash the tests:

```
+---------------------+
| Building spiffs.img |
+---------------------+
/net.js
/i2c.js
/ledc.js
/bootwifi.js
/start.js
/stream.js
/neopixels.js
/web/ide.html
/web/test.html
/web/selectAP.html
/web/ide.js
/Partitions.js
/uart_processor.js
/tests/test_serial_write_1.js
/tests/test_nvs_ints.js
/tests/test_neopixel_webserver.js
SPIFFS_write error(-10010): unknown
error adding file!
Error for adding content from tests!
```

The reason is that the filepath length is too long.  So, I have to move the `tests` dir out of `filesystem` before flashing.  I'm not sure how others managed to do this.

dashxdr, 2018-02-11:
test_neopixel_webserver.js rename to tst_neopixel_wbsrvr.js would fix it

boneskull, 2018-02-11:
...that isn't the only filename that's too long.  I can certainly shorten the filenames to flash, but I'm mainly curious if others were able to flash as-is.